VUnit/vunit #982: 1 test fals: Failed to find license notice in /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/build/lib/vunit/vhdl/osvvm/AlertLogPkg.vhd

yurivict, 2024-01-02:
```
========================================================================================= FAILURES ==========================================================================================
_______________________________________ TestLicense.test_that_a_valid_license_exists_in_source_files_and_that_global_licensing_information_is_correct _______________________________________

self = <tests.lint.test_license.TestLicense testMethod=test_that_a_valid_license_exists_in_source_files_and_that_global_licensing_information_is_correct>

    def test_that_a_valid_license_exists_in_source_files_and_that_global_licensing_information_is_correct(
        self,
    ):
        for file_name in find_licensed_files():
            code = ostools.read_file(file_name)
>           self._check_license(code, file_name)

tests/lint/test_license.py:51: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
tests/lint/test_license.py:67: in _check_license
    self.assertIsNotNone(match, "Failed to find license notice in %s" % file_name)
E   AssertionError: unexpectedly None : Failed to find license notice in /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/build/lib/vunit/vhdl/osvvm/AlertLogPkg.vhd
===================================================================================== warnings summary ======================================================================================
vunit/test/bench.py:29
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/bench.py:29: PytestCollectionWarning: cannot collect test class 'TestBench' because it has a __init__ constructor (from: tests/unit/test_test_bench.py)
    class TestBench(ConfigurationVisitor):

vunit/test/bench.py:276
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/bench.py:276: PytestCollectionWarning: cannot collect test class 'Test' because it has a __init__ constructor (from: tests/unit/test_test_bench.py)
    class Test(object):

vunit/test/bench_list.py:20
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/bench_list.py:20: PytestCollectionWarning: cannot collect test class 'TestBenchList' because it has a __init__ constructor (from: tests/unit/test_test_bench_list.py)
    class TestBenchList(object):

vunit/test/report.py:21
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/report.py:21: PytestCollectionWarning: cannot collect test class 'TestReport' because it has a __init__ constructor (from: tests/unit/test_test_report.py)
    class TestReport(object):

vunit/test/runner.py:27
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/runner.py:27: PytestCollectionWarning: cannot collect test class 'TestRunner' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestRunner(object):  # pylint: disable=too-many-instance-attributes

vunit/test/report.py:21
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/report.py:21: PytestCollectionWarning: cannot collect test class 'TestReport' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestReport(object):

vunit/test/list.py:14
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/list.py:14: PytestCollectionWarning: cannot collect test class 'TestList' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestList(object):

tests/unit/test_test_runner.py:217
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/tests/unit/test_test_runner.py:217: PytestCollectionWarning: cannot collect test class 'TestCaseMock' because it has a __init__ constructor (from: tests/unit/test_test_runner.py)
    class TestCaseMock(object):

vunit/test/suites.py:147
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/test/suites.py:147: PytestCollectionWarning: cannot collect test class 'TestRun' because it has a __init__ constructor (from: tests/unit/test_test_suites.py)
    class TestRun(object):

tests/unit/test_ui.py:1344
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/tests/unit/test_ui.py:1344: PytestCollectionWarning: cannot collect test class 'TestPreprocessor' because it has a __init__ constructor (from: tests/unit/test_ui.py)
    class TestPreprocessor(object):

tests/unit/test_ui.py:1357
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/tests/unit/test_ui.py:1357: PytestCollectionWarning: cannot collect test class 'TestPreprocessor2' because it has a __init__ constructor (from: tests/unit/test_ui.py)
    class TestPreprocessor2(Preprocessor):

tests/unit/test_ghdl_interface.py::TestGHDLInterface::test_compile_project_extra_flags
  /usr/ports/cad/py-vunit-hdl/work-py39/vunit-4.7.0/vunit/sim_if/ghdl.py:253: Warning: 'ghdl.flags' is deprecated and it will be removed in future releases; use 'ghdl.a_flags' instead
    warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
================================================================================== short test summary info ==================================================================================
SKIPPED [1] tests/acceptance/test_artificial.py:90: Only modelsim supports verilog
SKIPPED [1] tests/acceptance/test_artificial.py:121: Only modelsim supports verilog
SKIPPED [1] tests/acceptance/test_artificial.py:40: Only simulators with persistance functionality
SKIPPED [1] tests/acceptance/test_external_run_scripts.py:145: This simulator/backend does not support coverage
SKIPPED [1] tests/acceptance/test_external_run_scripts.py:62: Requires a Verilog simulator
SKIPPED [1] tests/acceptance/test_external_run_scripts.py:39: Requires a Verilog simulator
SKIPPED [1] tests/acceptance/test_external_run_scripts.py:58: Requires a Verilog simulator
============================================================ 1 failed, 558 passed, 7 skipped, 12 warnings in 2020.25s (0:33:40) =============================================================
```

Version: 4.7.0
Python-3.9
FreeBSD 14.0

LarsAsplund, 2024-02-25:
That file is part of third-party OSVVM which has another license that we ignore with these lines:

https://github.com/VUnit/vunit/blob/80d56a37021a6a7d81497161919784d2f4bb4191/tests/lint/test_license.py#L140-L144

For some reason it does not consider the file to be within the osvvm directory. Can you add some print statements for the two paths to figure out why?

LarsAsplund, 2026-11-02:
Closed for lack of activity.  Since OSVVM has been moved to the separate [vunit-osvvm package](https://github.com/VUnit/vunit-osvvm), any future issues should be reported in that repository.

closed 2026-11-02 by LarsAsplund