VUnit/vunit #983: test_runner_cleanup passes entry_gate after 1ms although it is locked

mkdev-0, 2024-01-03:
The simulation completes after 1ms. Logging with trace messages shows that _test_runner_cleanup_ passes the entry gate after 1ms without a lock being released. It works as expected if the delay is lower than 1ms.

---
`run.py`
``` run.py
from vunit import VUnit

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

# Optionally add VUnit's builtin HDL utilities for checking, logging, communication...
# See http://vunit.github.io/hdl_libraries.html.
vu.add_vhdl_builtins()
# or
# vu.add_verilog_builtins()

# Create library 'lib'
lib = vu.add_library("lib")

# Add all files ending in .vhd in current working directory to library
lib.add_source_files("tb*.vhd")

# Run vunit function
vu.main()
```
---
`tb_test.vhd`
``` tb_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library vunit_lib;
context vunit_lib.vunit_context;

entity tb_running_test_case is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_running_test_case is
  signal start_stimuli : event_t := new_event;
begin
  test_runner : process
  begin
    show_all(display_handler);
    test_runner_setup(runner, runner_cfg);

    notify(start_stimuli);

    test_runner_cleanup(runner);
  end process;

  stimuli_generator: process is
    constant key : key_t := get_entry_key(test_runner_cleanup);
    variable test_logger : logger_t := get_logger("test_logger");
  begin
    wait until is_active(start_stimuli);
    lock(runner, key, test_logger);

    info("Applying stimuli for scenario A - start");
    wait for 2 ms;
    info("Applying stimuli for scenario A - end");

    unlock(runner, key, test_logger);
  end process;

end architecture;
```
---
log when running `python run.py -v`
```
               0 fs - runner               -   TRACE - Entering test runner setup phase.
               0 fs - runner               -   TRACE - Passed test runner setup phase entry gate.
               0 fs - runner               -   TRACE - Passed test runner setup phase exit gate.
               0 fs - runner               -   TRACE - Entering test suite setup phase.
               0 fs - runner               -   TRACE - Passed test suite setup phase entry gate.
               0 fs - test_logger          -   TRACE - Locked test runner cleanup phase entry gate.
               0 fs - default              -    INFO - Applying stimuli for scenario A - start
               0 fs - runner               -   TRACE - Entering test runner cleanup phase.
               0 fs - runner               -   TRACE - Halting on test runner cleanup phase entry gate.
   1000000000000 fs - runner               -   TRACE - Passed test runner cleanup phase entry gate.
   1000000000000 fs - runner               -   TRACE - Passed test runner cleanup phase exit gate.
   1000000000000 fs - runner               -   TRACE - Entering test runner exit phase.
```

vunit-hdl 4.7.0
Python 3.11.6
archlinux

Jogt, 2024-02-06:
I encountered the same problem with a testbench. After some research in the VUnit code the problem seems to be the following:
In the [run library](https://github.com/VUnit/vunit/blob/ab56d3ff7aa80e88d0879c2f0be32321b19e529b/vunit/vhdl/run/src/run.vhd), inside the `test_runner_cleanup` procedure it attempts to enter the gate with the `entry_gate` procedure.
But here is the following wait statement:
  `wait on runner until not entry_is_locked(runner_state, get_phase(runner_state)) for max_locked_time;`
This constant `max_locked_time` is defined in [run_types.vhd](https://github.com/VUnit/vunit/blob/ab56d3ff7aa80e88d0879c2f0be32321b19e529b/vunit/vhdl/run/src/run_types.vhd#L17) with the value of 1 ms.

After I changed this value to a higher value (1 hr) the testbench behaves as expected.

This is the only occurrence of this constant and there is no function to modify or mention of this anywhere.
So I think this needs to be added to the documentation (https://vunit.github.io/blog/2023_04_01_vunit_phases.html#phase-gate-locks) and also there should be a method to change/disable this default value. After all, thats what `test_runner_watchdog` is for, or am I wrong?

LarsAsplund, 2024-02-25:
This feels like some relic from the past and I can't even remember why it's there. Probably because the watchdog wasn't present in early non-open source versions of VUnit. I think I will simply remove the timeout.

commit 80d56a37 (2024-02-25): Fixes #983
closed 2024-02-25 by LarsAsplund