VUnit/vunit #984: Support for merging coverage results with nvc

rpatel3001, 2024-01-03:
Merging coverage reports is [supported by nvc](https://www.nickg.me.uk/nvc/manual.html#Coverage_processing_options) but trying it in my run.py gives me `RuntimeError: This simulator does not support merging coverage`.

I was able to do it manually with:

```
nvc -c --merge=output.covdb input1.covdb input2.covdb input3.covdb
```

Blebowski, 2024-06-25:
Hi,

@rpatel3001 If this is still an issue in NVC (only one covdb in the generated report), could you please open a bug report there ?

tasgomes, 2024-08-21:
Hi everyone, is there any update on this?

rpatel3001, 2024-08-22:
i was just playing with nvc briefly, I haven't used it in a while. i don't know if either the vunit bug or nvc bug were fixed.