VUnit/vunit #999: Feature Request: real_array_t similar to integer_array_t

abaebae, 2024-03-11:
Hi, we use the fixed_pkg for a lot of signal processing stuff. We read the input data from csv and write the result back to csv and verify with post_check functions. Functionality similar to the integer_array_pkg would reduce the overhead of converting to integer and vice versa.
I would like to work on a draft for this, unless this is not a useful idea.

nselvara, 2024-03-21:
I think making a generic type package would solve the problem.

daedadu, 2024-03-21:
I would like something like this as well, I'm currently doing some signal processing with sfixed vhdl types and casting to signed lets me run into the 32 bit signed problem...

LarsAsplund, 2024-03-30:
This is related to #521