VUnit/vunit #1009: VUnit testcases fail if the testcase name contains a colon (SystemVerilog)

PossenigM, 2024-04-17:
I noticed that the VUnit testcases in a SystemVerilog testbench will always fail if their testcase name contains a colon. I attached a minimal example including a simple run.py script and a minimal testbench with two testcases. I tested this on two separate windows machines and got the same results on both pcs. Is this a bug in vunit or is something else wrong here?

[MinimalExample.zip](https://github.com/VUnit/vunit/files/15006799/MinimalExample.zip)

```python
from vunit import VUnit

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

vu.add_verilog_builtins()

# Create library 'lib'
lib = vu.add_library("lib")

# Add all files ending in .sv in current working directory to library
lib.add_source_files("*.sv")

# Run vunit function
vu.main()
```

```v
`include "vunit_defines.svh"

module tb_Test;
   `TEST_SUITE begin
        // Note: Do not place any code here (unless you are debugging
        // VUnit internals).

        `TEST_SUITE_SETUP begin
            $display("Running test suite setup code");
        end

        `TEST_CASE_SETUP begin
            $display("Running test case setup code");
        end

        `TEST_CASE("Test: Fail") begin
            
            // This testcase always fails because there 
            // is a colon in its name

        end

        `TEST_CASE("Test Pass") begin

            // This testcase will pass
            // Note: There is no colon in its name

        end

        `TEST_CASE_CLEANUP begin
            // This section will run after the end of a test case. In
            // many cases this section will not be needed.
            $display("Cleaning up after a test case");
        end

        `TEST_SUITE_CLEANUP begin
            // This section will run last before the TEST_SUITE block
            // exits. In many cases this section will not be needed.
            $display("Cleaning up after running the complete test suite");
        end
   end;

   // The watchdog macro is optional, but recommended. If present, it
   // must not be placed inside any initial or always-block.
   `WATCHDOG(1ns);
endmodule
```

LarsAsplund, 2024-05-28:
I will have to look closer why that is. Could also be simulator dependent. Which one are you using?

PossenigM, 2024-05-28:
I am using _Questa - Intel FPGA Starter Edition 2021.2_