VUnit/vunit #1010: How to specify test case(s) to run after creating a VUnit instance?

cjmeyer, 2024-04-23:
Is there a way to create the VUnit object and then specify the test case(s) to run? What I would like to do is create an instance of `VUnit` , call `add_library()`, `add_source_files()`, etc, and then specify the test case(s) I would like to run just before, or when, I call `main()`. It would maybe look something like:

```
vu = VUnit.from_args( ... )
vu.add_vhdl_builtins()

lib = vu.add_library('my_lib')
lib.add_sources(['src/*.vhd', 'tb/*.vhd'])

vu.main('my_lib.my_module.test_1')
```

My use case is using composable pytest fixtures to configure an instance of `VUnit` and then in the test case call `main()` (or equivalent) on the `VUnit` instance to actually build and run the simulation. One pytest fixture would create the base `VUnit` instance, and then successive fixtures would create libraries, add sources, set generics, etc, until the main test function called `main()`.

LarsAsplund, 2024-05-28:
What you can do is to modify the arguments in the script. https://vunit.github.io/py/ui.html#adding-custom-command-line-arguments explains how to add custom arguments and in that process the arguments are exposed such that you can modify them before creating the `VUnit` instance. You can use that approach but skipping the step where you actually add a custom argument

LarsAsplund, 2024-06-25:
@cjmeyer Did that answer your question, i.e. can this issue be closed?