VUnit/vunit #1021: Issue with using encrypted Vivado IP in Vunit

AN-SEOKHYEN, 2024-05-28:
Hi,
I am currently facing an issue while attempting to use encrypted Vivado IP with Vunit. Despite compiling the simulation libraries in Vivado with the Riviera-PRO simulator and adding the relevant libraries using add_external_library in my Vunitrun.py script, I am encountering the following error: "COMP96 ERROR COMP96_0078: Error in encrypted code."

Is there anything additional that needs to be done to use encrypted Vivado IP with Vunit? Any insights or suggestions would be greatly appreciated.

Thank you for your help.

LarsAsplund, 2024-05-28:
No, that should be sufficient and it sounds that the build from Vivado is somehow incompatible with Riviera-PRO. If you were to compile the IP with Riviera-PRO directly without involving VUnit, you would probably get the same result. Worth a try though.

AN-SEOKHYEN, 2024-05-28:
It seems like the issue has been resolved. It appears that using *.vhd in the add_source_files syntax inadvertently caused errors during the compilation process in VUnit by including encrypted VHDL files like 'fifo_generator_v13_2_vhsyn_rfs.vhd'.

If I understand correctly, to utilize the FIFO, it should suffice to add only the fifo_generator_0.vhd file using add_source_files. Of course, remember to include the compiled library as well.

For now, it seems resolved, but if there's any misunderstanding on my part, please let me know. Thank you!

LarsAsplund, 2024-05-28:
Yes, VUnit will not parse files to try to figure out if they are encrypted or not. It will simply call  the simulator to compile the file and in this case it will fail. Just keep these files out of the directory with your unencrypted files and you should be fine

closed 2024-05-28 by LarsAsplund