VUnit/vunit #1026: No warning of erroneous architecture when adding configurations

sigmuha, 2024-06-05:
When adding a generic to a testbench with the "add_config" method, no warning is given to the user if the architecture of the testbench is not correct. 

I experienced this when writing "t_dummy" instead of "tb_dummy" for the entity of the architecture of a testbench, where the entity is called "tb_dummy".  This caused vunit not to create a default configuration, leading to the following error when "add_config" was called:
```
File "C:\Users\dummy\AppData\Roaming\Python\Python312\site-packages\vunit\configuration.py", line 265, in add_config
    config = configs[DEFAULT_NAME].copy()
             ~~~~~~~^^^^^^^^^^^^^^
KeyError: None
```
What caused this error was not obvious at all and took some time to find what caused it. Spelling the architecture correctly fixed the issue. To remedy this, the shown error could be ignored, and continue with the compilation of all design and testbench files, which would highlight the misspelling. 

It is possible that different kinds of errors in the testbench file also can cause this, and not just a misspelling of the entity in the architecture declaration.

LarsAsplund, 2024-06-17:
Can you provide an [MRE](https://en.wikipedia.org/wiki/Minimal_reproducible_example). I can't seem to reproduce this so maybe I misunderstood your code.

sigmuha, 2024-06-24:
Sorry for the late reply.

[vunit_issue.tar.gz](https://github.com/user-attachments/files/15957588/vunit_issue.tar.gz) contains two files, run.py and test.vhd. 

Doing ```python run.py``` the same error as written above is reproduced. This happens, because a generic is attempted to be added, when no architecture for the testbench in test.vhd exists. In this case the architecture is misspelled. The user is not notified of this error.

LarsAsplund, 2024-06-25:
This should be fixed now.

commit 1678b89e (2024-06-24): Fixes #1026.
commit 1bad6790 (2024-06-25): Fixes #1026.
closed 2024-06-25 by LarsAsplund