VUnit/vunit #1029: [Question] ModelSim reports an error regarding VHDL-2019 package

nselvara, 2024-06-15:
If I try to run my Python script. Even though, I've set the env variable `VUNIT_VHDL_STANDARD` to 2008, ModelSim issues this error: `\.venv\Lib\site-packages\vunit\vhdl\logging\src\location_pkg-body-2019p.vhd(12) (vcom-1136) Unknown identifier "call_path_vector_ptr".`

I don't know how to exclude this 2019 pkg.

LarsAsplund, 2024-06-15:
Not sure what's happening. Call path requires both VHDL 2019 and a simulator supporting call paths. Modelsim is not. Without any more information I can only recommend a clean compile and running with --log-level=debug. Then post the output here.

nselvara, 2024-06-16:
Actually, before I post the log, I wanted to just say that I also got an error regarding `VendorCovApiPkg_Aldec`:

`C:\intelFPGA\23.1std\questa_fse\win64\vcom -quiet -modelsimini ".\vunit_out\modelsim\modelsim.ini" -2008 -work vunit_library ".\.venv\Lib\site-packages\vunit\vhdl\osvvm\VendorCovApiPkg_Aldec.vhd"`

=== Command output: ===
`** Error: .\.venv\Lib\site-packages\vunit\vhdl\osvvm\VendorCovApiPkg_Aldec.vhd(89): Function "VendorCovPointCreate" declared at line 52 has no body.`

I think it has to do with the `builtins.py`:
https://github.com/VUnit/vunit/blob/315adae5bd5b6e1648e4a2ee52dc7f91d934a540/vunit/builtins.py#L172-L173

I just replaced the empty body with the same definition of `VendorCovApiPkg.vhd`. With that I could fix the problem. I don't know why the aldec version is empty resp. even taken in first place.

nselvara, 2024-06-16:
After checking my run.py, I finally found my error. 
My VHDL files were located in `ip` folder, however, the directory that I passed on had `+ '/../'`, thus, it resulted to: ip_directory + '/../', which led to the inclusion of the .venv library as well. 😭
But thank you for your help.

closed 2024-06-16 by nselvara