VUnit/vunit #1031: [Possible Bug/Clarification] package xpm.vcomponents is not found

nselvara, 2024-06-18:
Hey there, I'm encountering a weird problem. I'm trying to simulate the `xpm.vcomponents.xpm_fifo_sync`.
The thing is, if I define in my code the library loading and the instantiation as this, it doesn't work.

```vhdl
library xpm;
-- NOTE: It seems to be that VUnit is not able to find the xpm_fifo_sync in VCOMPONENTS pkg.
use xpm.vcomponents.all;
⋮
xpm_fifo_inst: xpm_fifo_sync
⋮
```

But if I define it like this, VUnit manages to find the Verilog file at least. The annoying thing is that the VHDL LS can't analyse Verilog files though, hence, I can't hover over the label and see the entity's definition though.

```vhdl
library xpm;
-- NOTE: It seems to be that VUnit is not able to find the xpm_fifo_sync in VCOMPONENTS pkg.
use xpm.all;
⋮
xpm_fifo_inst: entity xpm.xpm_fifo_sync
⋮
```

I'm using `Questa Sim - Intel FPGA Edition 2023.3`. I'm also loading the proper libraries as well:

- `-L unisim`
- `-L unisims_ver`
- `-L xpm`

Am I missing something?

nselvara, 2024-06-21:
I somehow managed to fix the issue. When I loaded `unisims_ver` library, even though I ran with clean param, VUnit still couldn't find it. What I had to do was, just to delete the entire `.venv` folder and recreate it again, then it worked.
I don't know why --clean param didn't work.

LarsAsplund, 2024-06-22:
Figuring out exactly what went wrong would require more information on your run script and the error messages you received.

However, since you manages to resolve the issue anyway, I think the issue can be closed.

closed 2024-06-22 by LarsAsplund

nselvara, 2024-06-22:
The error message was something like: `Couldn't find (elaborate?) expanded xpm.vcomponents package`. However, I think it's also not really worth to investigate. It seems to be that you need all of the 3 libraries to compile properly.