VUnit/vunit #1038: GHDL 4.1.0 mcode backend not recognized

n8tlarsen, 2024-07-08:
$ python run.py
ERROR - Could not detect known LLVM backend by parsing 'ghdl --version'
Expected to find one of dict_keys(['mcode code generator', 'llvm (\\d+\\.\\d+\\.\\d+ )?code generator', 'GCC (back-end|\\d+\\.\\d+\\.\\d+) code generator'])
== Output of 'ghdl --version'============================================================
GHDL 4.1.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

=========================================================================================
Traceback (most recent call last):
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 735, in main
    all_ok = self._main(post_run)
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 779, in _main
    all_ok = self._main_run(post_run)
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 804, in _main_run
    simulator_if = self._create_simulator_if()
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/ui/__init__.py", line 798, in _create_simulator_if
    return self._simulator_class.from_args(args=self._args, output_path=self._simulator_output_path)
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/sim_if/ghdl.py", line 77, in from_args
    backend=cls.determine_backend(prefix),
  File "/home/nathan/.local/lib/python3.10/site-packages/vunit/sim_if/ghdl.py", line 145, in determine_backend
    raise AssertionError("No known GHDL back-end could be detected from running 'ghdl --version'")
AssertionError: No known GHDL back-end could be detected from running 'ghdl --version'

LukasVik, 2024-07-09:
I believe this was fixed in https://github.com/VUnit/vunit/commit/f38c1e340c029e8ce5554ee966bee6335aca3992 and https://github.com/VUnit/vunit/pull/1005. But this is only on the master branch still, not in the release PyPI package. Looks to me that you are using VUnit installed from PyPI. One solution would be to install from the latest git:

python3 -m pip install vunit-hdl@git+https://github.com/VUnit/vunit.git

Or wait until the next release.

n8tlarsen, 2024-07-09:
Thanks! Installing from the Git repository (version 5.0.0.dev0) solved the issue.

closed 2024-07-09 by n8tlarsen

LarsAsplund, 2024-07-15:
FYI. I will start pushing release candidates in cases where we have useful updates but want to delay the next release for some reason. In this case we want to release all planned breaking features at the same time (v5.0.0) to limit the number of major releases. Release candidates can be installed from pypi using the [--pre](https://pip.pypa.io/en/stable/cli/pip_install/#install-pre) option.