VUnit/vunit #1045: Regression: exploded stack in log function when running with Acitve-HDL

jobtijhuis, 2024-07-24:
Something in the log functions seems to be causing the "Error: KERNEL_0060 Stack area is too small, please increase it by the -stack option." error in Active-HDL 15 (Student Edition). There really seems to be something wrong because increasing the stack size does not allow the testbench to run. I think something changed in the log functions that causes the stack to explode in size.

The regression seems to be introduced in this commit: [b69eac2](https://github.com/VUnit/vunit/commit/b69eac230c54236f0b1bae9c168ecbbc2d407a24)

# Minimal reproduceable example
```vhdl
library ieee;
context ieee.ieee_std_context;
use ieee.math_real.all;

library vunit_lib;
use vunit_lib.check_pkg.all;
use vunit_lib.run_types_pkg.all;
use vunit_lib.run_pkg.all;

entity axi_state_transmission_tb is
  generic (
    CLOCK_PERIOD : time   := 10 ns;
    TEST_TIMEOUT : time   := 1 us;
    RUNNER_CFG   : string := RUNNER_CFG_DEFAULT
  );
end entity;

architecture sim of axi_state_transmission_tb is

  signal clock    : std_ulogic := '0';
  signal n_areset : std_ulogic := '0';

begin

  test_runner_watchdog(runner, TEST_TIMEOUT);

  process
  begin
    test_runner_setup(runner, RUNNER_CFG);

    vunit_lib.logger_pkg.error("Failed something");

    test_runner_cleanup(runner);
  end process;

end architecture;
```

LarsAsplund, 2024-08-15:
I haven't been able to reproduce this problem but I recall seeing it at some point but it disappeared for unknown reasons. I suggest you start updating to the latest VUnit release (clone from Github or do pip install with the --pre option). I'm using Active-HDL 14.0.258.8726.

jobtijhuis, 2024-08-15:
I was using the latest version of master when I encountered this problem. I will try to reproduce it on my work pc with an actual license. Maybe there are bug fixes in Active-HDL that the student version doesn't have.

jobtijhuis, 2024-10-07:
I isolated the files that I was having a problem with and I was not able to reproduce it. It seemed to be a problem with Active-HDL and black boxes. This error seemed to be given instead of the actual problem, which was that there were still black box entities in my design.

closed 2024-10-07 by jobtijhuis