VUnit/vunit #1046: OSVVM update

c-thaler, 2024-08-02:
OSVVM is currently rev-tied to a revision that is over a year old. In the meantime, there was a nice feature added that comes in handy when implementing bus masters (DelayCoverageIDType in DelayCoveragePkg.vhd).
It would be nice to have the rev-tie get bumped to the latest stable commit.

LarsAsplund, 2024-08-08:
@c-thaler I've pushed an update but I have yet to test over all simulators before merging.

c-thaler, 2024-08-08:
I will go one with the branch until it is merged. Thanks a lot!

nickg, 2024-08-09:
> Looks like NVC fails to compile this OSVVM version. @nickg do you know anything about this?

This was reported and fixed in nickg/nvc#917. Can you update the CI to use the 1.13.1 release?

LarsAsplund, 2024-08-09:
Will do. Thanks!

LarsAsplund, 2024-08-09:
@nickg The latest NVC fixed the problem but there is a new ambiguity problem. I'm on my cell phone so I haven't been able to test any workaround. Is that really ambitious?

nickg, 2024-08-09:
No it's a bug, I'll fix it this weekend. Sorry about that.

nickg, 2024-08-11:
> No it's a bug, I'll fix it this weekend. Sorry about that.

This is fixed in the latest 1.13.2 release.

LarsAsplund, 2024-08-11:
Thanks @nickg, I will rerun the test once the new NVC image is uploaded.

LarsAsplund, 2026-11-02:
This is an old issue that is not relevant anymore,

closed 2026-11-02 by LarsAsplund