VUnit/vunit #1057: [FEATURE] Print/save commands outputs/logs from (ModelSim/QuestaSim/Questa Base)

nselvara, 2024-09-12:
Hey there, recently, I got a request to run `simstats` and `printsimstats` commands in ModelSim. I thought, it might be interesting to have a command that executes before or after each simulation run cycle and prints/saves the log output. I don't if other simulators have also similar commands. Currently, the output.txt in vunit_out folders contain the information. However, it'd be cool to kind of just print that particular info without analysing 200 files. 😅

LarsAsplund, 2024-09-19:
This feels like something that is too simulator-specific to be hardcoded into VUnit. What could be added though is an option run a user-defined TCL script after a simulation just like we can before the simulation starts. Such a script can call `simstats` and the output can be further processed by Python in a post-simulation hook if needed.

nselvara, 2024-09-25:
Yeah, I understand your concern. The solution that you've proposed, works also, so for me either way is good.