VUnit/vunit #1059: Check code quality in a continuous way

racodond, 2024-09-20:
Hi,

I analyzed the VHDL code from your VUnit project with Linty: https://oss.linty-services.com/dashboard?id=vunit&codeScope=overall

Do not get scared by the number of issues :-)
It's just to provide you with data to browse. I activated all the rules, some are mutually opposed, others are not properly configured with your requirements (such as naming conventions).

We offer Linty for free for open-source projects. I'd be glad to organize a call if you're interested in checking code quality in a continuous way through GitHub actions.
We also offer our VS Code extension for free: https://hdl-designer.linty-services.com/

I'm looking forward to hearing from you.

Thank you

Have a good day

David Racodon
CTO at Linty Services

LarsAsplund, 2024-09-20:
Sounds interesting. Is it possible to enable/disable rules with the free version? All of the 677 security issues are all of this type:

[image]

That is obviously a rule we would need to disable or at least have the opportunity define `vunit_lib` as a safe library.

Code that doesn't synthesize well seems to generate several thousands warnings. Since this is a verification library, such rules need to be disabled as well.

racodond, 2024-09-21:
Hi @LarsAsplund,

Many thanks for your feedback!

> Is it possible to enable/disable rules with the free version?Once the false positives are removed it will be easier to spot the real issues

All features are available in the free version for open-source projects.
Obviously, you can activate/deactivate rules, configure them to match your needs and requirements, etc.

> That is obviously a rule we would need to disable or at least have the opportunity define vunit_lib as a safe library.

I updated your quality profile (set of rules) accordingly : https://oss.linty-services.com/coding_rules?activation=true&q=allowed+librar&qprofile=b670cf0d-0348-42b6-9259-fed380d85bd7&open=vhdl%3AVHDL153
I ran another analysis and the related issues have been closed: https://oss.linty-services.com/dashboard?id=vunit&codeScope=overall

> Code that doesn't synthesize well seems to generate several thousands warnings. Since this is a verification library, such rules need to be disabled as well.

Rules can either apply to files that will be synthesized or simulation files or both.
You can set the `sonar.hdl.file.simulationPaths` property at https://oss.linty-services.com/project/settings?id=vunit&category=hdl to tell Linty what you consider as testbenches, verfication lib, etc. to remove those "false-positives".

> Once the false positives are removed it will be easier to spot the real issues

You can privately provide me with an email address through https://linty-services.com/contact.html and I can create an account for you on our cloud platform. Then, you'll be able to update the set of rules according to your needs and run analyses.
We can also schedule a video conf for a demo and properly configure your project together.

Have a good day

LarsAsplund, 2024-09-21:
@racodond Please create such an account and I can give it a try. You can use the email in my profile.