VUnit/vunit #1061: [Bug] `check_equal` and `check_true` raises error for values other than `'0'` and `'1'`

nselvara, 2024-09-25:
Hey, there I noticed something strange, while using `check_equal` or `check_true` for (un-)signed types.
It actually raises an error if we pass any value except `'0'` and `'1'` for `got` resp. `expected`.

```vhdl
-- This passes
check_equal(got => std_ulogic_vector'("U"), expected => std_ulogic_vector'("U"), msg => "test std_ulogic_vector");
-- These not
check_equal(got => signed'("U"), expected => signed'("U"), msg => "test signed");
check_equal(got => unsigned'("X"), expected => unsigned'("X"), msg => "test unsigned");
check_true(expr => unsigned'("Z") = unsigned'("Z"), msg => "test equality unsigned");
```

I checked the definition of the (un-)signed types and seems to be that those are just arrays of `std_ulogic` type:

```vhdl
type UNRESOLVED_UNSIGNED is array (NATURAL range <>) of STD_ULOGIC;
type UNRESOLVED_SIGNED is array (NATURAL range <>) of STD_ULOGIC;
```

I guess, equality check with values other than ***number*** don't pass for (un-)signed types in VHDL resp. excluded in LRM?

```vhdl
if got = expected then
```

LarsAsplund, 2024-09-25:
VHDL is not always obvious and consistent. However, it would only create more confusion if VUnit was to check equality in a way that is not compatible with the `=` operator

nselvara, 2024-09-26:
Yeah, I understand that totally. I think there's also this function `UNSIGNED_EQUAL` in numeric-std-body, which converts the (un-)signed types to `std_ulogic_vector` beforehand. I think maybe expanding (either add. option, another name or completely something else) the `check_equal`-procedure with this sort of conversion would be cool though. But I understand if it's not desired.

LarsAsplund, 2024-12-13:
Since these are rather subtle details in VHDL, I think one should be explicit when taking advantage of them. So explicitly cast `unsigned` parameters to `std_logic_vector` in the `check_equal` call if the `std_logic_vector` style of equality is what you want. And then add a comment why you're doing it 😄

closed 2024-12-13 by LarsAsplund

nselvara, 2024-12-16:
Ok yeah, I see, is also an approach and in the end introducing a procedure for that is also doubtful that it's going to be used by others. But thank you for your time.

LarsAsplund, 2024-12-16:
Yes, I think it will be used rather rarely and in the end we want readability, i.e. a procedure name that states that we are checking equality like `check_equal` but we are using a different definition of equality. Something like `check_equal_as_if_parameters_were_unsigned`. Explicitly casting the parameters to a normal `check_equal` call is just as easy and readers of the code would still wonder why this is needed so a comment is required in either case.