VUnit/vunit #1063: Publish patch for GHDL mcode JIT support

sweeneyal, 2024-09-29:
Hey all,

I see you already patched the mcode JIT support error on master, but not in a release. I understand you are waiting until you have all your breaking changes in order for v5.0.0. However, if the only thing blocking the use of GHDL mcode 4.1.0 is the change in the printout, could you release a patch version (e.g. v4.7.1) that supports the mcode JIT printout?

LarsAsplund, 2024-09-30:
Would a v5.0.0 pre-release work for you? Do pip install with the `--pre` option?

sweeneyal, 2024-09-30:
I forgot that was an option! Let me try that and get back to you.

sweeneyal, 2024-11-27:
I finally got around to using the prelease patches, took a little googling but if you're using a requirements file, include VUnit via:

`vunit_hdl>=5.0.0.dev3`

Thanks guys, this helped. Cheers!

closed 2024-11-27 by sweeneyal