VUnit/vunit #1068: [Bug] Package parsing misses dependencies when `use` clause is not used

SzymonHitachi, 2024-10-22:
Assume having a package in a file `pkg_a.vhd`:
```vhdl
package pkg_a is

    type my_type_t is (A, B, C)

end package pkg_a;
```

and a second package `pkg_b` in a file `pkg_b.vhd` which scopes an object from `pkg_a`

```vhdl
package pkg_b is

    alias mytype is work.pkg_a.mytype;

end package pkg_b;
```
When used in above way `VUnit` doesn't detect dependency between `pkg_b.vhd` and `pkg_a.vhd`. This is a valid use case in VHDL.

However if type is imported directly through `use` clause, the dependency is detected.

```vhdl
-- any of the below works
use work.pkg_a.all;
use work.pkg_a.mytype;

package pkg_b is

    alias mytype is work.pkg_a.mytype;

end package pkg_b;
```

LarsAsplund, 2024-10-22:
Yes, there are legal VHDL/SV constructs that VUnit can't detect. Our ambition has not been to create a complete parser as that would increase the analysis time and it would also be a major effort considering that HDLs are hard to parse. Our aim is to have a parser that covers the more common cases which leaves you with two options. Either write your VHDL in a "VUnit friendly" way (as you suggested) or use [ add_dependency_on`](https://vunit.github.io/py/vunit.html#vunit.ui.source.SourceFile.add_dependency_on) is you're not in control of how the RTL is written.

I also see a future where we could make use of an external open source parser that fits our purposes. A dedicated parser project may give us a more complete parser without sacrificing performance.

LarsAsplund, 2024-12-12:
See #1080

closed 2024-12-12 by LarsAsplund