VUnit/vunit #1069: [Bug] Multiline comment containing semicolon over port declaration breakes vhdl_parser.parse()

SzymonHitachi, 2024-10-22:
When parsing a comment in format
```vhdl
/* 
 COMMENT
 */
a : in std_logic;
```

E.g.
if `COMMENT` contains `:` e.g. if it's an url:
```vhdl
/* 
 https://support.xilinx.com
 */
a : in std_logic;
```
parser errors with
```python
  File "vunit\vunit\source_file.py", line 221, in __init__
    design_file = vhdl_parser.parse(self.name)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 36, in parse
    return cached(
           ^^^^^^^
  File "vunit\vunit\cached.py", line 45, in cached
    result = function(content)
             ^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 77, in parse
    entities=list(VHDLEntity.find(code)),
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 363, in find
    yield VHDLEntity.parse(sub_code[: match.end()])
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 386, in parse
    ports = cls._find_port_clause(code)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 450, in _find_port_clause
    return cls._parse_port_clause(code[match.start() : match.end() + closing_pos + match_semicolon.end()])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 548, in _parse_port_clause
    port_list.append(VHDLInterfaceElement.parse(interface_element, is_signal=True))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 667, in parse
    subtype_indication = VHDLSubtypeIndication.parse(interface_element_string.split(":")[1].strip())
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "vunit\vunit\vhdl_parser.py", line 612, in parse
    type_mark = subtype_indication_declaration.group("type_mark")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'group'
```

probably due to `interface_element_string.split(":")` which probably shouldn't be done in multiline comment block

LarsAsplund, 2024-10-22:
VUnit doesn't detect block comments. The reasoning for limiting its capabilities is the same as in #1068

SzymonHitachi, 2024-11-20:
Make sense. Would be a good idea to note it somewhere that `:` in multiline comments is not supported by VUnit, since I agree that fixing this might be problematic.