VUnit/vunit #1072: [Feature] Open a beta and/or alpha channel

nselvara, 2024-11-11:
Hey there, I guess, I write it here, maybe, there's more support for this. As discussed in this issue: https://github.com/VUnit/vunit/issues/1028.
Opening a beta/alpha channel for example on pip, where one can pull the latest commits, would reduce the waiting time for major releases and also helps companies that have a more narrow requirement.

LarsAsplund, 2024-11-13:
I've released some pre-releases which are available by calling `pip` with the `--pre` option, see https://pypi.org/project/vunit-hdl/#history and https://vunit.github.io/installing.html#using-the-python-package-manager

LarsAsplund, 2024-12-12:
@nselvara Does this solve your request?

nselvara, 2024-12-13:
@LarsAsplund, hi there, sorry, I forgot to reply.
Yeah, the pre-packages solved the issue, however, we didn't want to use the  `--pre` option for `pip`, as it would install also the pre-versions for the rest of the requirements.
Thus, we made a workaround to install only the particular version in `requirements.txt`: `vunit_hdl>=5.0.0.dev3`
With this one might restrict only some packages to deviate outside of release versions.

closed 2024-12-13 by nselvara