VUnit/vunit #1077: Unable to simulate a VHDL design using a SV testbench.

usman1515, 2024-12-07:
I am using Vunit for the first time and trying to integrate it into my project. I have a Vivado project with design made in VHDL and test benches made in System Verilog.
I have Vivado 2021.2 installed on my local machine along with other FOSS tools including GHDL and iverilog. However I am unable to compile my design of a simple `mux_2x1`.
Below attached is my `run.py` file along with `mux_2x1.vhd` and `tb_mux_2x1.sv`.

This is the error that I am getting.
> Compiling into vunit_lib: venv_vunit/lib/python3.13/site-packages/vunit/verilog/vunit_pkg.sv   ERROR - Unknown file type: systemverilog
failed
File type not supported by ghdl simulator
Compile failed

What exactly am i doing wrong here.
- can i do this in the first place?
- can i change the simulator or am i stuck with the default one.

### file: `run.py`
```
from pathlib import Path
from vunit import VUnit

prj_dir = Path(__file__).parent
print('prj_dir: ', prj_dir)

rtl_path = Path(__file__).parent / 'sample_space'
tb_path = Path(__file__).parent / 'sample_space'

print('dir_rtl: ', rtl_path)
print('dir_tb:  ', tb_path)

# create vunit instance py passing cmd line args
# vu = VUnit.from_argv()
vu = VUnit.from_argv(compile_builtins=False)

# add vunit builtin HDL utilities for checking, logging, communication
vu.add_vhdl_builtins()
vu.add_verilog_builtins()

# create library lib
lib = vu.add_library('lib_rtl').add_source_files(rtl_path / '*.vhd')
lib = vu.add_library('lib_tb').add_source_files(tb_path / '*.v')

# run vunit function
vu.main()

```
### file: `mux_2x1.vhd`
```
library IEEE;
use IEEE.STD_LOGIC_1164.all;
use IEEE.NUMERIC_STD.all;
use IEEE.STD_LOGIC_UNSIGNED.ALL;

entity mux_2x1 is
    port (
		in_data_0   : in std_logic_vector(64-1 downto 0);
        in_data_1   : in std_logic_vector(64-1 downto 0);
        in_sel      : in std_logic;
        out_data    : out std_logic_vector(64-1 downto 0)
	);
end mux_2x1;

architecture Behavioral of mux_2x1 is

begin

	process(in_data_0, in_data_1, in_sel)
    begin
        if in_sel = '0' then
            out_data <= in_data_0;
        else
            out_data <= in_data_1;
        end if;
    end process;

end Behavioral;
```
### file: `tb_mux_2x1.sv`
```
`timescale 1ns / 1ps
`include "vunit_defines.svh"

module tb_mux_2x1;
    
    parameter string runner_cfg = runner_cfg_default;

    // vunit test suit declaration
    `TEST_SUITE begin
        // NOTE: Do not place any code here

        `TEST_SUITE_SETUP begin
            $display("Running test case setup code");
            localparam T = 1;
            localparam DATA_WIDTH = 64;
            // localparam runner_cfg;

            // IO signals
            reg [DATA_WIDTH-1 : 0] in_data_0;
            reg [DATA_WIDTH-1 : 0] in_data_1;
            reg in_sel;
            wire [DATA_WIDTH-1 : 0] out_data;

            // entity instantiation
            mux_2x1 DUT_mux_2x1 (
                .in_data_0(in_data_0),
                .in_data_1(in_data_1),
                .in_sel(in_sel),
                .out_data(out_data)
            );
        end

        initial begin
            // provide random seed values
            in_data_0 = 64'h393b5fa5f7da2daa;
            in_data_1 = 64'h3f78e789e9de3edf;

            repeat(20) begin
                in_data_0 = in_data_0 << 1'b1;
                in_data_1 = in_data_1 << 1'b1;
                in_sel = $urandom_range(1'b0, 1'b1);
                #T;
            end

            #T;
            // $finish;
        end

    end

    `WATCHDOG(1ns);

endmodule

```

LarsAsplund, 2024-12-11:
GHDL only supports VHDL so you need to install a simulator supporting mixed-language simulations. For example ModelSim, Questa, Riviera-PRO, or Active-HDL.

usman1515, 2024-12-12:
noted. thanks

usman1515, 2024-12-12:
noted. And as of now Vivado mixed signal isn't supported.

LarsAsplund, 2024-12-12:
Correct. I would recommend to download the free [Intels Starter Edition](https://www.intel.com/content/www/us/en/software-kit/750368/modelsim-intel-fpgas-standard-edition-software-version-18-1.html) of ModelSim. I think it supports mixed-language simulations but I not sure.

closed 2024-12-13 by usman1515