VUnit/vunit #1083: Add option "add_help=False" when creating an ArgumentParser object.

PelleEikeberg, 2024-12-12:
Right now there is no option for creating an ArgumentParser object without the help option. ArgumentParser() allows for the `add_help=False` option. Adding this option to the __init__ of VUnitCLI would allow us to then use the resulting ArgumentParser object as a parent of other ArgumentParser() commands. 

The argument for this is when I want to have Vunit as part of a larger script, and therefore add other options to the --help tag.

Changing the lines (line 44-82 in vunit_cli.py) to add this would be simple and only add usability.

```
class VUnitCLI(object):
    """
    VUnit command line interface
    """

    def __init__(self, add_help_descr=True, description=None):
        """
        :param description: A custom short description of the command line tool
        """
        self.parser = _create_argument_parser(description, add_help_descr=add_help_descr)
        self.add_help_descr = add_help_descr

    def parse_args(self, argv=None):
        """
        Parse command line arguments

        :param argv: Use explicit argv instead of actual command line argument
        :returns: The parsed argument namespace object
        """
        return self.parser.parse_args(args=argv, add_help_descr=self.add_help_descr)

def _create_argument_parser(description=None, for_documentation=False, add_help_descr=True):
    """
    Create the argument parser

    :param description: A custom short description of the command line tool
    :param for_documentation: When used for user guide documentation
    :returns: The created :mod:`argparse` parser object
    """
    if description is None:
        description = f"VUnit command line tool version {version()!s}"

    if for_documentation:
        default_output_path = "./vunit_out"
    else:
        default_output_path = str(Path(os.getcwd()).resolve() / "vunit_out")

    parser = argparse.ArgumentParser(description=description, add_help=add_help_descr)
```

PelleEikeberg, 2024-12-12:
I tried to create a branch for pull-request, but it seems I don't have access to do this. I might be new to Git Hub and therefore missing some obvious way, but all the code requiring change is in the example code above.

LarsAsplund, 2024-12-12:
To create a PR you will have to fork the VUnit project first, see https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo

Regarding your use case. Have you looked at https://vunit.github.io/py/ui.html#adding-custom-command-line-arguments? That is how you would add custom options to the existing ones.