VUnit/vunit #1085: Vivado 2024.2 and VHDL 2008 files

dordijeb, 2024-12-17:
Hi,

My flow fails here:

`RuntimeError: Unsupported compile order file: sources_1/ip/ddc_dds/hdl/float_pkg.vhd`.

In compile_order.txt that file is listed as:
`dds_compiler_v6_0_26,VHDL 2008,sources_1/ip/ddc_dds/hdl/float_pkg.vhd`

Seems that the issue is caused when upgrading to Vivado 2024.2  and VHDL 2008 is not supported in vunit as compile order file.

LarsAsplund, 2024-12-17:
Seems like there have been updates in their format. The problem is these lines:

https://github.com/VUnit/vunit/blob/35d1aa6766477ee30254c65c18168e99f969491f/vunit/vivado/vivado.py#L95-L99

Add the following just before line 95:

``` python
if file_type.startswith("VHDL"):
    file_type = "VHDL"
```

Does that solve the problem?

Do you know if there is a specification for what file types that we can expect in such a file?

dordijeb, 2025-02-04:
That worked, thanks :)

I also had to change to vhdl-2008.

closed 2025-02-04 by dordijeb