VUnit/vunit #1096: Questa 3-step-flow

maierch1SICKAG, 2025-01-29:
Hi,

I am using version 5.0.0.dev5 to prepare our internal IP regression for use Questa 3-step-flow.
The simulations works so far and show a noticeable improvement in simulation time.

Unfortunately, I don't get the visualizer gui simulation working with Questa 2024.2. on WSL2 / Ubuntu 22.04

When I run the script with 
  `python3 run.py --gui --debug=visualizer`

I got

`** Error: Mutually exclusive options -gui, -visualizer given.
** Error: Simulation exiting.`

Did I miss something?

LarsAsplund, 2025-01-29:
I'm running 2024.3 under Windows and that works. However, there are sometimes differences between versions and I don't think the `-gui` option is required. Try removing/commenting out this line:

https://github.com/VUnit/vunit/blob/ce51561f57dee4346373da5c4af2831d4e7dd22d/vunit/sim_if/vsim_simulator_mixin.py#L325

Does it make a difference? It will break the GUI mode without the Visualizer so a real fix involves more changes.

maierch1SICKAG, 2025-01-30:
Hi Lars,

this does the trick. Now, visualizer is starting up.
I can crosscheck the behaviour with 2024.3 in the next days.

LarsAsplund, 2025-01-30:
I pushed a more proper fixed that you can try.

maierch1SICKAG, 2025-02-03:
Thanks, the fix works with 2024.2.

Now, you can easily switch between Visualizer and legacy Gui.

LarsAsplund, 2025-02-04:
Great, then I will close this issue!

closed 2025-02-04 by LarsAsplund