VUnit/vunit #1099: nvc: Waveform files are only generated with --gui option

tmeissner, 2025-02-25:
It seems that when using nvc as simulator, the waveforms set with `--viewer-fmt` format are only generated when using the `--gui` option of the Python runner. With ghdl, this isn't the case, waveforms are generated regardless of the `--gui` option.

I think, the problem is in nvc.py `simulate()` method:

First, `wave_file` is only set to a value if `_gui` is set:
https://github.com/VUnit/vunit/blob/acf7e7f294f9b1b11dc664c2f145b9cb95906532/vunit/sim_if/nvc.py#L260-L265

Later if `wave_file` is set, the nvc `--wave` is added to the command line:
https://github.com/VUnit/vunit/blob/acf7e7f294f9b1b11dc664c2f145b9cb95906532/vunit/sim_if/nvc.py#L295-L296

In ghdl.py, creation of the waveform export command line option is only dependent on `_viewer_fmt`, but not on `_gui`:

https://github.com/VUnit/vunit/blob/acf7e7f294f9b1b11dc664c2f145b9cb95906532/vunit/sim_if/ghdl.py#L368-L375

https://github.com/VUnit/vunit/blob/acf7e7f294f9b1b11dc664c2f145b9cb95906532/vunit/sim_if/ghdl.py#L327-L333

LarsAsplund, 2025-03-03:
@oscargus Do you have any comments on this?

oscargus, 2025-03-04:
There is https://github.com/VUnit/vunit/pull/1042 which solves it. (There is also a bit of discussions/rambling in #1003 about the need for it.)

My thought has been to add a command to generate the waveforms in a more "non-surprising" way than specifying `--viewer-fmt`, but got wound up in what to call it...

`--waveform(s)` or `--wave(s)`, but then one should probably have called `--viewer-fmt` `--wave-fmt` or something. And there I got stuck as I had just renamed it once... The idea is, to clarify, to specify that waveforms should be generated (but the gui should not be opened, so calling it something related to viewer seems confusing).

Coming back to it, I guess it can make sense to have a flag like that (`--wave*`) and then add aliases for `--viewer-fmt` that matches the selected name.

If you agree @LarsAsplund , and can make an executive decision on the naming, I should be able to provide a PR within a week or so.

oscargus, 2025-03-05:
I added #1101 which unifies things.