VUnit/vunit #1107: Wishbone master doesn't handle `wait_until_idle()` properly

abyszuk, 2025-03-31:
Hi,

Turns out that wishbone master BFM doesn't handle the `wait_until_idle()` call properly. I think it's missing a wait for clock edge.
The missing `wait until rising_edge(clk)` code is here:
https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/wishbone_master.vhd#L94C1-L97C16
but you can see this wait present just a few lines above:
https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/wishbone_master.vhd#L68C1-L71C16

LarsAsplund, 2025-04-01:
@abyszuk Can you clarify a bit more what you see and what you expect?

abyszuk, 2025-04-01:
In my VHDL testbench I have the following sequence of events:
```
write();  -- control register, start
wait_until_idle();
read();  -- status register, check busy bit
```
In Wishbone this should cause CYC signal to drop for at least one clock cycle. However, that's not what I see on a waveform:

[image]
You can see that CYC stays high.
However, when I zoom in to delta cycles I can see that WB master internal `cycle` signal drops momentarily, but goes back before the next clock edge.

[image]

When I compare the master code in those two places I mentioned, it's quite clear that someone forgot to add `wait until rising_edge(clk)`.