VUnit/vunit #1108: Test list empty when test suite included from separate file

kstripp, 2025-04-18:
My organization is only just starting to use VUnit, and many reviewers, as well as potential future developers, may not be familiar with it.  As a result, I'd like to set things up so that someone unfamiliar with VUnit can simply open up the simulator (Questa / Visualizer in our case), and run sims directly from the console. 

To that end, I tried moving all of the test suite directives in to a separate package file that is conditionally included.

There is a corresponding package file (not included in my MWE) that has the conventional `initial` blocks to allow simulation without using VUnit.  This seems to work fine.

However, when the `TEST_SUITE` and `TEST_CASE` directives are brought in via `include`, I get the error `** Error: Found no "" test case`  (full eror below). Omitting the `include` gives the expected warning that `runner_cfg` is missing.  This seems to imply that VUnit can partially(?) find the test cases in the included file, but somehow not completely.

I included a minimum example of this behavior.  As the names imply, `run_working.py` shows correct test discovery and execution, `run_fails.py` shows this behavior (bug? user error?).

Plan B would be making fully separate testbenches for VUnit to use, but I was hoping to avoid that.

Thanks for all the incredible work in producing this library!  So far I like it much more than my previous regression solutions (including Questa VRM)

[include_testcase_bug.tar.gz](https://github.com/user-attachments/files/19806566/include_testcase_bug.tar.gz)

Full error message (with sanitized paths)
```
# ** Error: Found no "" test case
#    Time: 0 ps  Scope: vunit_pkg.test_runner.loop File: XXXX/env/lib/python3.13/site-packages/vunit/verilog/vunit_pkg.sv Line: 164
# Break in Function vunit_pkg/test_runner::loop at XXXX/env/lib/python3.13/site-packages/vunit/verilog/vunit_pkg.sv line 164
# Stopped at XXXX/env/lib/python3.13/site-packages/vunit/verilog/vunit_pkg.sv line 164
# 
# Test Run Failed!
# 
# Stack trace result from 'tb' command
# XXXX/env/lib/python3.13/site-packages/vunit/verilog/vunit_pkg.sv 164 return [address 0x7fefe8336964] Function vunit_pkg/test_runner::loop
# called from  XXXX/vunit_cfg_pkg.sv 1 return [address 0x7fefe833874e]  
# 
# 
# Surrounding code from 'see' command
#   159 :                   // A test suite without explicit test cases
#   160 :                   cleanup();
#   161 :                   return 0;
#   162 :                end
#   163 :                else if (!found) begin
# ->164 :                   $error("Found no \"%s\" test case", test_cases_to_run[j]);
#   165 :                   $stop(1);
#   166 :                   return 0;
#   167 :                end
#   168 :             end
# 
fail (P=0 S=0 F=1 T=1) lib.core_tb.all (0.7 seconds)

==== Summary ===========================
fail lib.core_tb.all (0.7 seconds)
========================================
pass 0 of 1
fail 1 of 1
========================================
Total time was 0.7 seconds
Elapsed time was 0.7 seconds
========================================
Some failed!

```

LarsAsplund, 2025-04-19:
When splitting testbench and test cases you need to tell VUnit where to look. See if [this](https://vunit.github.io/py/vunit.html#vunit.ui.testbench.TestBench.scan_tests_from_file) helps:

```
lib.add_source_file("fails_tb.sv")
tb = lib.test_bench("core_tb")
tb.scan_tests_from_file("vunit_cfg_pkg.sv")
```

kstripp, 2025-04-20:
Thank you!  That worked perfectly.  Clearly I should have read more of the fine manual....

closed 2025-04-20 by kstripp