VUnit/vunit #1109: Output the command that sends to the simulator

yangyt96, 2025-04-18:
Can you add an option to output the command which is send to the simulator, for example:

```
vlog -v2k5 -incr -l xpm -l xil_defaultlib -work xil_defaultlib  "+incdir+../../../../project.ip_user_files/ipstatic" \
	"../../../../project.gen/sources_1/ip/test_ip/test_ip.v" \
	"../../../../project.gen/sources_1/ip/test_ip/test_ip.v" \

vcom -93 -incr -work xil_defaultlib  \
	"../../../../../../tb/mmcm/mmcm_tb.vhd" \

vlog -work xil_defaultlib "glbl.v"

```

It only pop-ups the command when there is compilation/simulation error, however I think it is necessary for debugging purpose to check whether the command line is correct or not.

For example, some specific command for Vivado IP core simulation is required, this will result different simulation output. (The compilation and simulation can pass without error, but some special option is required to add to the simulator and I would like to know whether it is added correctly.)

LarsAsplund, 2025-04-19:
Have you tried running with the  `--log-level=debug` and `-v` options? Does it help?

yangyt96, 2025-04-21:
It does not help, the `--log-level=debug` and `-v` only gives me the message of

```
DEBUG - Adding source file <souce-path> to library <library-name>
DEBUG - Adding primary design unit (package) <entity-name>
DEBUG - Adding secondary design unit (package body) for package <entity-name>
DEBUG - The file <file-path> has no components
```

This only tells me the logging step and a lot of info are covered.

I would like to see the command line which is sent to the simulator as the example I have shown above:
```
vlog -v2k5 -incr -l xpm -l xil_defaultlib -work xil_defaultlib  "+incdir+../../../../project.ip_user_files/ipstatic" \
	"../../../../project.gen/sources_1/ip/test_ip/test_ip.v" \
	"../../../../project.gen/sources_1/ip/test_ip/test_ip.v" \

vcom -93 -incr -work xil_defaultlib  \
	"../../../../../../tb/mmcm/mmcm_tb.vhd" \

vlog -work xil_defaultlib "glbl.v"
```

LarsAsplund, 2025-07-08:
I will add this.