VUnit/vunit #1119: [BUG] Simulation Discrepancy Between VUnit v5.0.0-dev.4 and v5.0.0-dev.5 with Intel's altclklock IP

nselvara, 2025-05-30:
Hi there,
I've encountered a version-dependent simulation failure when running a testbench involving Intel's `altclklock` PLL primitive using VUnit.

---

### Summary

- Using VUnit `v5.0.0-dev.4`: simulation **passes**
- Using VUnit `v5.0.0-dev.5`: simulation **hangs and fails**, showing:

  ```
  ps Warning: Duty Cycle Violation
  ERROR - pll_locked - Got 0. Expected 1.
  ```
  
- **Quartus version**: `22.4`
- **Primitive**: `altclklock` from `altera_mf_components.vhd` (`eda/sim_lib` path)

---

### Steps to Reproduce

1. Create a VUnit testbench using the Intel `altclklock` primitive (example below).
2. Install VUnit version `v5.0.0-dev.4` -> test passes.
3. Install VUnit version `v5.0.0-dev.5` -> test fails with duty cycle violations and lock failure.
4. Even when `clean=True` is passed to `run.py`, dev.5 fails, while dev.4 passes.
5. Cloning the project fresh also fails under dev.5.
6. Only difference observed between environments (confirmed via `pip freeze`) was the VUnit version.

---

### Diff Link

[[Compare dev.4...dev.5](https://github.com/VUnit/vunit/compare/v5.0.0-dev.4...v5.0.0-dev.5)](https://github.com/VUnit/vunit/compare/v5.0.0-dev.4...v5.0.0-dev.5)

---

### Possibly Related Change

The [[release notes](https://vunit.github.io/release_notes.html)](https://vunit.github.io/release_notes.html) for dev.5 state:

> *Provided a unique base seed to every simulation.*

However, our testbench **does not use randomisation**, so this alone shouldn't affect results.
After analysing the actual diff, we suspect the root cause might be:

#### Retry Logic

- `optimize` calls are now wrapped in `_execute_with_retries(...)`, defaulting to 3 attempts.
- For timing-sensitive IPs like PLLs, retries could disturb initialisation and locking.

#### Verilog Time Precision Flags Changed

```diff
- "-t ps"
+ "-t", "ps"
```

- A subtle change, but possibly influencing how ModelSim GUI interprets resolution.

#### Generic Load Flag Format Updated

```diff
- -gentity:generic=value
+ -gentity/generic=value
```

- Switched from colon to slash. Could affect IPs where `inclock_period` or `valid_lock_cycles` are passed via generics.
- Potential side effect if older or patched versions of ModelSim/Visualizer are used.

---

### Notably

- All **other testbenches** in our suite still pass successfully under both versions.
- Only the `altclklock`-based test fails.

---

### Minimal Reproducible Example (MRE)

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library vunit_lib;
context vunit_lib.vunit_context;

library altera_mf;
use altera_mf.altera_mf_components.all;

entity tb_minimal_pll is
    generic (
        runner_cfg : string := runner_cfg_default
    );
end entity;

architecture test of tb_minimal_pll is
    constant SYS_CLK_PERIOD: time := 10 ns;
    constant HALF_CLK_PERIOD: time := SYS_CLK_PERIOD / 2;
    constant SIMULATION_TIME: time := 10 us;

    constant REF_CLK_PERIOD_PS: positive := 10000;
    constant CLK_MULTIPLICATION: positive := 8;
    constant CLK_DIVISION: positive := 1;
    constant PHASE_SHIFT_PS: natural := 0;
    constant VALID_LOCK_CYCLES: natural := 5;
    constant INVALID_LOCK_CYCLES: natural := 5;

    signal sys_clk_in: std_logic := '0';
    signal pll_clk_en: std_logic := '0';
    signal pll_clk_out: std_logic;
    signal pll_locked: std_logic;
begin
    clk_gen: process
    begin
        sys_clk_in <= '0';
        loop
            wait for HALF_CLK_PERIOD;
            sys_clk_in <= not sys_clk_in;
        end loop;
    end process;

    test_runner_watchdog(runner, SIMULATION_TIME);

    main: process
    begin
        test_runner_setup(runner, runner_cfg);

        while test_suite loop
            if run("test_pll_locks") then
                pll_clk_en <= '1';

                for i in 1 to 100 loop
                    if pll_locked = '1' then
                        exit;
                    end if;
                    wait for SYS_CLK_PERIOD;
                end loop;

                check_equal(got => pll_locked, expected => '1', msg => "pll_locked");

                for i in 1 to 5 loop
                    wait until pll_clk_out'event;
                end loop;

                info("PLL clock is toggling and locked");
            end if;
        end loop;

        test_runner_cleanup(runner);
    end process;

    pll: altclklock
        generic map (
            inclock_period => REF_CLK_PERIOD_PS,
            inclock_settings => "UNUSED",
            valid_lock_cycles => VALID_LOCK_CYCLES,
            invalid_lock_cycles => INVALID_LOCK_CYCLES,
            operation_mode => "NORMAL",
            clock0_boost => CLK_MULTIPLICATION,
            clock0_divide => CLK_DIVISION,
            clock0_settings => "UNUSED",
            clock0_time_delay => "0",
            outclock_phase_shift => PHASE_SHIFT_PS,
            intended_device_family => "Arria 10",
            lpm_type => "altclklock"
        )
        port map (
            inclock => sys_clk_in,
            inclocken => pll_clk_en,
            fbin => pll_clk_out,
            clock0 => pll_clk_out,
            locked => pll_locked
        );
end architecture;
```

---

### Request

Could you please confirm if any of the changes in `dev.5` (retry logic, flag syntax, or generic format) might influence simulation startup or generic propagation - particularly for clocking IPs like `altclklock`?

Thanks a lot in advance - If any help is needed, please don't hesitate to ask me.

LarsAsplund, 2025-07-08:
Have you tried to figure out what the duty cycle warning means? I don't think any the changes you mentions should be the direct cause of the bug of the bug but I've seen stranger things happening.

LarsAsplund, 2025-07-08:
I can't really say what is going wrong but it sounds like a simulator quirk that you may never find the real cause of. I would simply continue with the ps resolution. After all, I would expect a PLL model to be in the ps domain.

I also noticed that your solution includes more than changing the resolution. For example, switching off optimization. Simulator failures due to optimizations are fairly common in my experience. Sometimes a rewrite of the code to something functionally equivalent is the solution but sometimes you just have to give up and switch it off.

nselvara, 2025-07-09:
# Follow-up: Simulation time with `-t ps` vs `-t 1ns`

Interestingly, I just ran some comparison tests and the simulation time actually went **down** when switching from `-t 1ns` to `-t ps`:

```
Before (`-t 1ns`)
================================================================================================================
Total time was 3 min 23.6 s
Elapsed time was 3 min 24 s
================================================================================================================

After (`-t ps`)
================================================================================================================
Total time was 3 min 11.9 s
Elapsed time was 3 min 12.4 s
================================================================================================================
```

I'd have assumed finer resolution would lengthen it, but it seems the opposite happened, probably because it avoids the cycle violation lock or related overhead.

So the `-t ps` workaround not only fixes the issue but actually gives us a slight speed-up too.  
My company can definitely live with that!

Thanks again for the input, I’ll stick with this for now.