VUnit/vunit #1120: question: support to extended identifiers in the VUnit VHDL Parser?

rafaelnp, 2025-06-08:
Hello,

When using PeakRDL-regblock-vhdl in tandem with VUnit I have the following error executing the run.py script:
(although it does not interrupts the testbench execution)

```sh
ython3 run.py
Traceback (most recent call last):
File "/home/rnp/src/vunit/vunit/source_file.py", line 223, in __init__
design_file = vhdl_parser.parse(self.name)
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 36, in parse
return cached(
"CachedVHDLParser.parse",
...<3 lines>...
database=self._database,
)
File "/home/rnp/src/vunit/vunit/cached.py", line 34, in cached
result = function(content)
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 80, in parse
packages=list(VHDLPackage.find(code)),
~~~~~~~~~~~~~~~~^^^^^^
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 266, in find
result = list(cls._find_normal_packages(code))
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 246, in _find_normal_packages
yield cls.parse(sub_code[: match.end()])
~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 278, in parse
list(VHDLRecordType.find(code)),
~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 775, in find
subtype_indication = VHDLSubtypeIndication.parse(identifier_list_and_subtype_indication[1].strip())
File "/home/rnp/src/vunit/vunit/vhdl_parser.py", line 613, in parse
type_mark = subtype_indication_declaration.group("type_mark")
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'group'
ERROR - Failed to parse /home/rnp/tmp/peakhdl/PeakRDL/examples/regblock/atxmega_spi_pkg.vhd
WARNING - /home/rnp/tmp/peakhdl/PeakRDL/examples/regblock/atxmega_spi.vhd: failed to find a primary design unit 'atxmega_spi_pkg' in library 'atxmega_spi'
Re-compile not needed

WARNING - /home/rnp/tmp/peakhdl/PeakRDL/examples/regblock/atxmega_spi.vhd: failed to find a primary design unit 'atxmega_spi_pkg' in library 'atxmega_spi'
(21:12:13) Starting atxmega_spi.atxmega_spi_tb.all
Output file: /home/rnp/tmp/peakhdl/PeakRDL/examples/regblock/vunit_out/test_output/atxmega_spi.atxmega_spi_tb.all_5b209e0e58bd1b61262df2eb937a48cfa70dcc11/output.txt
pass (P=1 S=0 F=0 T=1) atxmega_spi.atxmega_spi_tb.all (0.8 s)

==== Summary ==========================================
pass atxmega_spi.atxmega_spi_tb.all (0.8 s)
=======================================================
pass 1 of 1
=======================================================
Total time was 0.8 s
Elapsed time was 0.8 s
=======================================================
All passed!
```

run.py:

```python
from pathlib import Path
from vunit import VUnit

VU = VUnit.from_argv(vhdl_standard="2008")
VU.add_vhdl_builtins()

VU.add_library("atxmega_spi").add_source_files(["*.vhd"])
script_path = Path(__file__).resolve().parent
tb = VU.library("atxmega_spi").test_bench("atxmega_spi_tb")

VU.main(
```

atxmega_spi_tb.vhd:

```vhdl
library ieee;
context ieee.ieee_std_context;

library vunit_lib;
context vunit_lib.vunit_context;

entity atxmega_spi_tb is
	generic(
		runner_cfg : string
	);
end entity atxmega_spi_tb;

architecture rtl of atxmega_spi_tb is

begin

	proc_name: process
	begin
		test_runner_setup(runner, runner_cfg);

		wait for 500 ns;

		test_runner_cleanup(runner);

	end process proc_name;
end architecture rtl;

```

It looks like, the VUnit VHDL parser does not support extended attributes. Could you please take a look and confirm it is the case?

OS: Linux 6.14.9-arch1-1 
Python: 3.13.3
VUnit: v5.0.0-dev.6 
PeakRDL-regblock-vhdl: 1.0.0.3
Questa: 22.01

Commands to reproduce the error:
(save the run.py and atxmega_spi_tb.vhd to PeakRDL/examples)

```sh 
pipx install peakrdl --include-deps
pipx inject peakrdl peakrdl-regblock-vhdl
git clone git@github.com:SystemRDL/PeakRDL.git
cd PeakRDL/examples
python run.py
```

rafaelnp, 2025-11-07:
My apologies for the delayed reply. The `atxmega_spi_pkg.vhd` file, generated by running `peakrdl  regblock-vhdl atxmega_spi.rdl -o regblock --cpuif axi4-lite`: 

```vhdl 
-- Generated by PeakRDL-regblock-vhdl - A free and open-source VHDL generator
--  https://github.com/SystemRDL/PeakRDL-regblock-vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package atxmega_spi_pkg is
    constant ATXMEGA_SPI_DATA_WIDTH : integer := 8;
    constant ATXMEGA_SPI_MIN_ADDR_WIDTH : integer := 2;
    constant ATXMEGA_SPI_SIZE : integer := 4;

    

    type \atxmega_spi.CTRL.MASTER_in_t\ is record
        next_q : std_logic;
        we : std_logic;
    end record;

    type \atxmega_spi.CTRL_in_t\ is record
        MASTER : \atxmega_spi.CTRL.MASTER_in_t\;
    end record;

    type \atxmega_spi.STATUS.WRCOL_in_t\ is record
        next_q : std_logic;
        we : std_logic;
    end record;

    type \atxmega_spi.STATUS.IF_in_t\ is record
        next_q : std_logic;
        we : std_logic;
    end record;

    type \atxmega_spi.STATUS_in_t\ is record
        WRCOL : \atxmega_spi.STATUS.WRCOL_in_t\;
        \IF\ : \atxmega_spi.STATUS.IF_in_t\;
    end record;

    type \atxmega_spi.DATA.RDATA_in_t\ is record
        next_q : std_logic_vector(7 downto 0);
    end record;

    type \atxmega_spi.DATA_in_t\ is record
        RDATA : \atxmega_spi.DATA.RDATA_in_t\;
    end record;

    type atxmega_spi_in_t is record
        CTRL : \atxmega_spi.CTRL_in_t\;
        STATUS : \atxmega_spi.STATUS_in_t\;
        DATA : \atxmega_spi.DATA_in_t\;
    end record;

    type \atxmega_spi.CTRL.PRESCALER_out_t\ is record
        value : std_logic_vector(1 downto 0);
    end record;

    type \atxmega_spi.CTRL.MODE_out_t\ is record
        value : std_logic_vector(1 downto 0);
    end record;

    type \atxmega_spi.CTRL.MASTER_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.CTRL.DORD_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.CTRL.ENABLE_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.CTRL.CLK2X_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.CTRL_out_t\ is record
        PRESCALER : \atxmega_spi.CTRL.PRESCALER_out_t\;
        MODE : \atxmega_spi.CTRL.MODE_out_t\;
        MASTER : \atxmega_spi.CTRL.MASTER_out_t\;
        DORD : \atxmega_spi.CTRL.DORD_out_t\;
        ENABLE : \atxmega_spi.CTRL.ENABLE_out_t\;
        CLK2X : \atxmega_spi.CTRL.CLK2X_out_t\;
    end record;

    type \atxmega_spi.INTCTRL.INTLVL_out_t\ is record
        value : std_logic_vector(1 downto 0);
    end record;

    type \atxmega_spi.INTCTRL_out_t\ is record
        INTLVL : \atxmega_spi.INTCTRL.INTLVL_out_t\;
    end record;

    type \atxmega_spi.STATUS.WRCOL_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.STATUS.IF_out_t\ is record
        value : std_logic;
    end record;

    type \atxmega_spi.STATUS_out_t\ is record
        WRCOL : \atxmega_spi.STATUS.WRCOL_out_t\;
        \IF\ : \atxmega_spi.STATUS.IF_out_t\;
    end record;

    type \atxmega_spi.DATA.WDATA_out_t\ is record
        value : std_logic_vector(7 downto 0);
    end record;

    type \atxmega_spi.DATA_out_t\ is record
        WDATA : \atxmega_spi.DATA.WDATA_out_t\;
    end record;

    type atxmega_spi_out_t is record
        CTRL : \atxmega_spi.CTRL_out_t\;
        INTCTRL : \atxmega_spi.INTCTRL_out_t\;
        STATUS : \atxmega_spi.STATUS_out_t\;
        DATA : \atxmega_spi.DATA_out_t\;
    end record;
end package;
```

LarsAsplund, 2026-02-26:
This shouldn't be too hard to support. Before adding, I will have to evaluate the performance impact for parsing. Extended identifiers are not commonly used so it must not have a significant negative performance impact on the majority that are using basic identifiers.