VUnit/vunit #1129: Questa 3-step-flow issue

tasgomes, 2025-08-07:
Hi @LarsAsplund,

I think I found a problem with the 3-step-flow. 

I have this use-case where I simulate a Vivado managed IP (.xci file), which uses a verilog simulation model. As is the case with Verilog and AMD primitives, one needs to load the [glbl.v](https://adaptivesupport.amd.com/s/article/6537?language=en_US) module into the simulation environment, using:

`lib.set_sim_option("modelsim.vsim_flags", ["glbl_ver.glbl"])`

I noticed that if the `modelsim.three_step_flow` is `false` then the simulation runs fine.
But if the `modelsim.three_step_flow` is `true` I get the following error during run-time:

```
# vsim opt_libtbtesttbtb -modelsimini C:/temp/VUnit_issue/fw/sim/vunit_out/questa/modelsim/modelsim.ini -work lib_tb -quiet -t 1ps -onfinish stop -voptargs="+acc" glbl_ver.glbl -L vunit_lib -L osvvm -L unisim -L glbl_ver -L xpm -L blk_mem_gen_v8_4_9 -L lib_tb -wlf C:/temp/VUnit_issue/fw/sim/vunit_out/questa/test_output/lib_tb.test_tb.test_rw_b132e4d5875a0122bf10f89f74fbb34ea6f8c5b8/modelsim/vsim.wlf -g/test_tb/runner_cfg="active python runner : true,enabled_test_cases : test_rw,output path : C::/temp/VUnit_issue/fw/sim/vunit_out/questa/test_output/lib_tb.test_tb.test_rw_b132e4d5875a0122bf10f89f74fbb34ea6f8c5b8/,seed : 74324863e2cb91db,tb path : C::/temp/VUnit_issue/fw/tb/test/,use_color : true"
# Start time: 12:18:11 on Aug 07,2025
# ** Warning: (vsim-3824) Optimized (opt_libtbtesttbtb) and unoptimized (glbl) top-level design units are
# simulated together. Unoptimized design units will not be optimized.
# ** Error (suppressible): (vsim-12110) All optimizations are disabled because the -novopt option is in effect. This will cause your simulation to run very slowly. If you are using this switch to preserve visibility for Debug or PLI features, please see the User's Manual section on Preserving Object Visibility with vopt. -novopt option is now deprecated and will be removed in future releases.
# Error loading design
Error loading design
# End time: 12:18:11 on Aug 07,2025, Elapsed time: 0:00:00
# Errors: 1, Warnings: 1
```

It looks like the `glbl` sim flag is not passed correctly to `vopt` when `three_step_flow` is enabled (?)

VUnit Version: `v5.0.0-dev.4` (but it also happens with `v5.0.0-dev.6`)

I created a minimal setup to reproduce issue: [VUnit_issue.zip](https://github.com/user-attachments/files/21663500/VUnit_issue.zip).

- Extract the ZIP file
- Change directory to `fw/` folder
- Execute the command: `python3 -u ./sim/run.py -o ./sim/vunit_out/questa --simulator questa --simulator-path C:/intelFPGA_pro/24.3/questa_fe/win64 --vendor-lib ./sim/sim_lib/questa` (Set the simulator path to your questa installation folder)
  - Test case will pass.
- Go to `fw/sim/tb/test/run.py` and enable `three_step_flow` in line 66.
  - Test case will fail.

tasgomes, 2025-08-11:
Thanks for hint. Actually I am the one manually compiling them since I am using an Intel-variant Questa that is not supported by Vivado.

I will see if I manage to optimized them and report back the result.

LarsAsplund, 2025-08-24:
`Bad flag from vopt_args` is something that we print on any failure but that is rarely the cause. The dot has been discussed in the past but personally, I've never had to do that. Do you have a reproduceable example on that can be shared?

Generally speaking the optimization fails now and then without any clear indications on what is going on. I have a client running Questa Sim and also a vendor version of Questa. Some code works on one version but not the other, with or without three-step-flow, and with or without extra flags to suppress optimizations. Most often, the solution to make it work has been to reduce the amount of optimization using flags such as `+acc`, `-O`, and `-noprotectopt`

harry-commin-enclustra, 2025-08-30:
@tasgomes I hit exactly the same issue with Questa FE 2023.4, and I also moved my `glbl` from `"modelsim.vsim_flags"` to `"modelsim.vopt_flags"` (after finding https://github.com/VUnit/vunit/issues/679#issuecomment-2581920708). That resolved the initial issue, but I also continued to see different errors.

In my case, the solution was simply that I also needed to `-suppress` some errors via `vopt_flags`, which were previously being suppressed via `vsim_flags`. Specifically, with Questa FE, it panics if the design is too large:

```
** Error (suppressible): (vopt-14408) Intel  FPGA Edition recommended capacity is 5000 non-OEM instances. There are 108406 non OEM instances. Expect performance to be severely impacted.
```

Suppressing this everywhere fixed the issue for me:
```
    lib.set_compile_option("modelsim.vcom_flags", ["-suppress", "14408"])
    lib.set_sim_option("modelsim.vsim_flags", ["-suppress", "14408"])
    lib.set_sim_option("modelsim.vopt_flags", ["-suppress", "14408"])
```